byuccl/bfasst #113: bfasst can't run without rapidwright

jgoeders, 2023-02-21:
Since the physical netlist script imports from rapidwright, it's not currently possible to use run_design.py without having rapidwright installed.  It would be nice to have a workaround to remove this dependency.

dallinjdahl, 2023-02-21:
We could do something like this: https://stackoverflow.com/a/3496790

reillymck, 2023-02-21:
Also, this is something that would be appropriate for the __init__.py file.

dallinjdahl, 2023-02-21:
We could also refactor the flows file and load the relevant one when asked.  Then it would automatically report an error if we needed rapidwright, but if not, it would run just fine...

reillymck, 2023-02-21:
I really like Dallin's idea. The way I understand it, is we would move the imports so that they are inside the relevant function calls. So the import happens when the function is called, not when the run_design is loaded. That is what you are suggesting correct?

jgoeders, 2023-02-21:
That sounds like the best approach to me.

dallinjdahl, 2023-02-21:
That's basically the gist :)

closed 2023-03-16 by jgoeders