byuccl/bfasst #115: Reverse_bit_tool attribute error

JakeEdvenson, 2023-03-01:
I've tried running several flows using `python3 scripts/run_design.py examples/byu/alu xilinx_yosys_impl` (replacing the last part with other flow names) and whenever it reaches `status = xray_rev(design, build_dir, flow_args)` it throws an error at line 21 in scripts/bfasst/reverse_bit/base.py where it says `AttributeError: 'collections.defaultdict' object has no attribute 'read'`. I'm running this on the current commit from main in a virtual environment. Any thoughts?

reillymck, 2023-03-02:
This is an interesting bug. The flow function for xilinx_yosys_impl (I added the print statements):
```
def flow_xilinx_yosys_impl(design, flow_args, build_dir):
    """Vivado synthesis and implementation, reverse with xray, compare with yosys"""
    print(f"Full flow args: {flow_args}")
    status = vivado_synth(design, build_dir, flow_args)
    print(f"Full flow args1: {flow_args}")
    status = vivado_impl(design, build_dir, flow_args)

    # Run X-ray and fasm2bel
    print(f"Full flow args2: {flow_args}")
    status = xray_rev(design, build_dir, flow_args)

    status = yosys_cmp(design, build_dir, flow_args)
    return status
```
I also added print statements in the base classes for synth, impl, and reverse_bit to print the flow_args argument passed to the constructor. This is the print I get:
```
Full flow args: defaultdict(<class 'str'>, {<ToolType.SYNTH: 0>: '', <ToolType.IMPL: 1>: '', <ToolType.MAP: 2>: '', <ToolType.CMP: 3>: '', <ToolType.REVERSE: 5>: ''})
Synth Base flow args: 
Synthesis already run 
Full flow args1: defaultdict(<class 'str'>, {<ToolType.SYNTH: 0>: '', <ToolType.IMPL: 1>: '', <ToolType.MAP: 2>: '', <ToolType.CMP: 3>: '', <ToolType.REVERSE: 5>: ''})
Impl flow args: 
Implementation already run 
Full flow args2: defaultdict(<class 'str'>, {<ToolType.SYNTH: 0>: '', <ToolType.IMPL: 1>: '', <ToolType.MAP: 2>: '', <ToolType.CMP: 3>: '', <ToolType.REVERSE: 5>: ''})
Reverse flow args: defaultdict(<class 'str'>, {<ToolType.SYNTH: 0>: '', <ToolType.IMPL: 1>: '', <ToolType.MAP: 2>: '', <ToolType.CMP: 3>: '', <ToolType.REVERSE: 5>: ''})
Traceback (most recent call last):
  File "/home/reilly/equiv/bfasst/scripts/run_design.py", line 87, in <module>
    main()
  File "/home/reilly/equiv/bfasst/scripts/run_design.py", line 83, in main
    run_design(design_path, args.flow, args.error_flow, flow_args=flow_args)
  File "/home/reilly/equiv/bfasst/scripts/run_design.py", line 31, in run_design
    status = run_flow(design, flow, flow_args, build_dir)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/reilly/equiv/bfasst/scripts/bfasst/flows.py", line 106, in run_flow
    return flow_fcn(design, flow_args, build_dir)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/reilly/equiv/bfasst/scripts/bfasst/flows.py", line 206, in flow_xilinx_yosys_impl
    status = xray_rev(design, build_dir, flow_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/reilly/equiv/bfasst/scripts/bfasst/tool_wrappers.py", line 124, in xray_rev
    reverse_bit_tool = XRayReverseBitTool(build_dir, flow_args)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/reilly/equiv/bfasst/scripts/bfasst/reverse_bit/xray.py", line 19, in __init__
    super().__init__(cwd, flow_args)
  File "/home/reilly/equiv/bfasst/scripts/bfasst/reverse_bit/base.py", line 22, in __init__
    self.args = parser.parse_args(shlex.split(flow_args))
                                  ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/reilly/.conda/envs/eqv/lib/python3.11/shlex.py", line 315, in split
    return list(lex)
           ^^^^^^^^^
  File "/home/reilly/.conda/envs/eqv/lib/python3.11/shlex.py", line 300, in __next__
    token = self.get_token()
            ^^^^^^^^^^^^^^^^
  File "/home/reilly/.conda/envs/eqv/lib/python3.11/shlex.py", line 109, in get_token
    raw = self.read_token()
          ^^^^^^^^^^^^^^^^^
  File "/home/reilly/.conda/envs/eqv/lib/python3.11/shlex.py", line 140, in read_token
    nextchar = self.instream.read(1)
               ^^^^^^^^^^^^^^^^^^
AttributeError: 'collections.defaultdict' object has no attribute 'read'
```
So actually, only reverse_bit is getting the correct data structure (which fails the shlex parse). The real question is why isn't synth/impl getting the dictionary structure.  Even more interesting is when I remove the default argument of `""` from the constructors for synth/impl base and vivado_synth/vivado_impl classes, the print is the same. So really there is two bugs here. We should not be calling shlex.split on a dictionary and we need to see why the flow args are not making it to synth/impl
@jgoeders

reillymck, 2023-03-02:
Update the problem is in the tool wrapper. Forgot about those... Fix is in my pull request

reillymck, 2023-03-02:
The current state of the tool wrappers will currently cause the opt flows to fail. I'm not sure what flow_args we want to pass to those tools.

jgoeders, 2023-03-07:
Is this fixed with #116?

jgoeders, 2023-03-07:
@reillymck

reillymck, 2023-03-07:
Yes.

closed 2023-03-07 by reillymck