byuccl/bfasst #127: Statuses aren't printed correctly in run_experiment.py

jgoeders, 2023-03-18:
When using run_experiment.py, Status objects are printed incorrect (it shows the Status type twice):

```
(.venv) jgoeders@Jeff:~/bfasst$ python scripts/run_experiment.py experiments/tests/phy_netlist_structural_cmp.yaml 
Running 16 designs 
byu/Sout_reg                Transform Error (Transform Error (Unhandled primitive INBUF_EN))
byu/UpDownButtonCount       Transform Error (Transform Error (Unhandled primitive INBUF_EN))
byu/alu                     Transform Error (Transform Error (Unhandled primitive INBUF_EN))
byu/buttoncount             Transform Error (Transform Error (Unhandled primitive CFF))
byu/calc                    Transform Error (Transform Error (Unhandled primitive INBUF_EN))
byu/debouncer               Transform Error (Transform Error (Unhandled primitive BFF))
byu/lightWave               Transform Error (Transform Error (Unhandled primitive OUTBUF))
byu/mux8to1                 Transform Error (Transform Error (Unhandled primitive PAD))
byu/oneshot                 Transform Error (Transform Error (Unhandled primitive PAD))
byu/regfile                 Transform Error (Transform Error (Unhandled primitive AFF))
byu/riscvSimpleDatapath     Transform Error (Transform Error (Unhandled primitive PAD))
byu/rx                      Transform Error (Transform Error (Unhandled primitive PAD))
byu/seven_segment           Transform Error (Transform Error (Unhandled primitive PAD))
byu/shiftReg                Transform Error (Transform Error (Unhandled primitive A5FF))
byu/stopwatch               Transform Error (Transform Error (Unhandled primitive DFF))
byu/tx                      Transform Error (Transform Error (Unhandled primitive INBUF_EN))
```

reillymck, 2023-05-01:
Also, rapidwright output is printed to the terminal instead of to log files:
<pre>
ooc/fixed_point_sqrt              
ooc/hight                         Transform Error (Transform Error (Unhandled primitive FDCE))
Running: fm_3d_co (0:05:34) graphiti (0:05:25) jpegenco (0:04:47) lfsr_ran (0:04:12) m32632 (0:02:30) md5_pipe (0:02:13) <b>...INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable
RW_QUIET_MESSAGE=1.</b>
ooc/md5_pipelined                 !! Synth Error (!! Synth Error)
Running: fm_3d_co (0:05:43) graphiti (0:05:34) jpegenco (0:04:56) lfsr_ran (0:04:21) m32632 (0:02:39) median (0:01:37) ...INFO: <b>Finished building uncommon Wire->Node cache</b>
ooc/graphiti                      
ooc/lfsr_randgen                  Equivalent
</pre>
Rapidwright print is in bold. The rapidwright text should be printed to the log file of its associated process.

jgoeders, 2023-05-03:
I thought I fixed this with this line: 
https://github.com/byuccl/bfasst/blob/71d6542f8575ae3bcbfe1b5de1e620bccc350e61/bfasst/transform/xilinx_phys_netlist.py#L121

...but I guess it's not fully working.

jgoeders, 2023-05-30:
It's likely something to do with the Status class and how we are converting to string.  https://github.com/byuccl/bfasst/blob/e52ea7cd2a1279899039d3612afe442483d91ecc/bfasst/status.py#L113.  I'm not sure if the issue is what we are doing in run_experiment.py or in status.py?  Have you reproduced the issue yet?  Run something using run_experiment.py, and perhaps change a module to always return an error and look at how the output is formatted.  Here is an example of returning an error status: https://github.com/byuccl/bfasst/blob/084282e5d163f835847e480a2559f8b40ec4f307/bfasst/synth/vivado.py#L153

jgoeders, 2023-06-01:
Fixed by #194.

@reillymck Do you know how to trigger the rapidwright message?  We should open a separate issue for that.

reillymck, 2023-06-02:
I tried reproducing with a fresh rw install, but I could not. I think we can just leave it be until we encounter it again. The fresh rw install wouldn't have built the wire->node cache so it should trigger the print, but I got nothing when I ran the ooc designs.

jgoeders, 2023-06-15:
@KeenanRileyFaulkner  It looks like the PR to fix this issue broke other places where the status is printed.

For example, if `structural.py` returns an error status (`return Status(CompareStatus.COULD_NOT_MAP)`), the following is printed:

```
Traceback (most recent call last):
  File "/home/jgoeders/bfasst/scripts/run_design.py", line 87, in <module>
    main()
  File "/home/jgoeders/bfasst/scripts/run_design.py", line 83, in main
    run_design(Path(args.design_path), args.flow, args.error_flow, flow_args=flow_args)
  File "/home/jgoeders/bfasst/scripts/run_design.py", line 37, in run_design
    status = run_flow(design, flow, flow_args, build_dir)
  File "/home/jgoeders/bfasst/bfasst/flows.py", line 119, in run_flow
    return flow_fcn(design, flow_args, build_dir)
  File "/home/jgoeders/bfasst/bfasst/flows.py", line 206, in flow_xilinx_phys_netlist_cmp
    status = structural_cmp(
  File "/home/jgoeders/bfasst/bfasst/tool_wrappers.py", line 63, in structural_cmp
    return struct_cmp.compare_netlists()
  File "/home/jgoeders/bfasst/bfasst/compare/structural.py", line 108, in compare_netlists
    return Status(CompareStatus.COULD_NOT_MAP)
  File "/home/jgoeders/bfasst/bfasst/status.py", line 111, in __init__
    raise BfasstException(status, f": {self.msg}")
bfasst.status.BfasstException: :
```

I believe it used to print the associated error message, which would be "Could not map" (https://github.com/byuccl/bfasst/blob/f4423cae065ef16f13841856df648677493d2372/bfasst/status.py#L86)

KeenanRileyFaulkner, 2023-06-15:
Ok. I'll take another look later today

KeenanRileyFaulkner, 2023-06-15:
Should be solved now. There was a circular dependency in the way that run_experiment created Statuses from BfasstExceptions that had been created by Statuses.

KeenanRileyFaulkner, 2023-06-16:
Closed by #218

closed 2023-06-16 by KeenanRileyFaulkner