byuccl/bfasst #154: CI fails with java.lang.RuntimeException: ERROR: Failed to reliably download file

jgoeders, 2023-03-31:
The CI fails about half the time with an error downloading a file.
For example:
```
java.lang.RuntimeException: ERROR: Failed to reliably download file: data/devices/artix7/xc7a200t_db.dat
```

This happens while running experiments, not setting up the environment, which is unexpected, because the Rapidwright files are downloaded during environment setup.  Not sure why files would be downloaded during the experiment run.

reillymck, 2023-05-04:
It appears that rapidwright only downloads the database for parts when you use them for the first time instead of downloading all of them at install...

jgoeders, 2023-05-04:
Can we force it to download a specific part and retry on failure?

reillymck, 2023-05-04:
I made an adjustment in ooc in the run_rapidwright function in xilinx_phys_netlist:
```
        try:
            self.rw_design = Design.readCheckpoint(
                design.xilinx_impl_checkpoint_path, design.impl_edif_path
            )
        except jpype.JException as exc:
            error = str(exc)
            if "Failed to reliably download file" in error:
                self.rw_design = Design.readCheckpoint(
                design.xilinx_impl_checkpoint_path, design.impl_edif_path
            )
            else:
                raise RapidwrightException(str(exc))
```
This will try and second time. What do you think?

reillymck, 2023-05-04:
Or do we want to edit the make file so it downloads the database during install? Not sure how to do that, but I can look into it.

jgoeders, 2023-05-04:
Yeah this looks like a good start.

I agree during setup would be better. Or in https://github.com/byuccl/bfasst/blob/main/bfasst/jpype_jvm.py

jgoeders, 2023-05-18:
https://github.com/Xilinx/RapidWright/issues/669

jgoeders, 2023-05-18:
The issue was due to running jobs in parallel -- the files were trying to be downloaded by multiple instances of rapidwright simultaneously.

PR #182 (2023-05-18): Get rapidwright data files on install
Fixes #154
closed 2023-05-18 by reillymck