byuccl/bfasst #171: VHDL OOC designs fail to compile due to missing packages

reillymck, 2023-05-01:
```
ooc/ahbctrl                       !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'stdlib' not found in library 'grlib' [bfasst/designs/ooc/ahbctrl/ahbctrl.vhd:31]))
ooc/ahbjtag                       !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'amba' not found in library 'grlib' [bfasst/designs/ooc/ahbjtag/ahbjtag.vhd:30]))
ooc/ahbuart                       !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'amba' not found in library 'grlib' [bfasst/designs/ooc/ahbuart/ahbuart.vhd:30]))
ooc/apbctrl                       !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/apbctrl/config.vhd:6]))
ooc/dsu3                          !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/dsu3/config.vhd:6]))
ooc/grethm                        !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/grethm/config.vhd:6]))
ooc/irqmp                         !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/irqmp/config.vhd:6]))
ooc/leon3s                        !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/leon3s/config.vhd:6]))
ooc/mctrl                         !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/mctrl/config.vhd:6]))
ooc/spimctrl                      !! Synth Error (!! Synth Error ([Synth 8-4169] error in use clause: package 'gencomp' not found in library 'techmap' [bfasst/designs/ooc/spimctrl/config.vhd:6]))
```

jgoeders, 2023-05-03:
I think this used to work, so something changed at some point and they broke.

KeenanRileyFaulkner, 2023-06-06:
Hey so I talked with Reilly and it appears that the vendor for these designs released them with their own Makefiles to compile them. The makefile takes into account the vendor/board you are compiling for. So it is possible that these files were downloaded for lattice and they worked great with that, but are not building for xilinx since the author has vendor-specific build files. We are not sure how to best integrate their specific build system into the bfasst build system, and if we wanted to include these designs, maybe they deserve their own directory in designs like GRLIB (the distributor). Reilly is wondering if it is worth it to edit bfasst to incorporate this vendor's build system. Regardless, we could remove these designs from OOC, and then add a recipe in the Makefile like `make grlib` that would curl/extract the designs from https://gaisler.com/index.php/downloads/leongrlib into the designs directory. Then we could either leave it up to the user to build the designs or we could create a grlib flow that builds those specific designs correctly by calling their makefiles. The vendor also specifies specific versions of Vivado for the design/board you are targeting. We took the leon processor build script targeting the xc7a100t board and tried building it with Vivado 2022.2 (the script is written for 2018) and it failed. Reilly thinks that it just seems like a lot to get these specific designs running. What do you think? @reillymck @jgoeders

jgoeders, 2023-06-06:
This design (LEON3) is the only "real" large design we have, so I would like to get it working.

Can you point me to an example in the HDL where they are using lattice primitives?

Also, what is relying on Vivado 2018?  Maybe we can talk more about this in person...

reillymck, 2023-06-06:
Good to know about the Leon3. I don't know if there is lattice specific primitives per say but I went to the distributor's website and they include their own makefiles to build their project. Keenan and I download the tarball and it contained different directory's for a variety of boards that you could target for the Leon. We went into the digilent artix7 directory and tried to run the makefile targeting xilinx and it failed. But the README did specify Vivado 2018.2 and we used 2022.2. Vivado is giving weird library errors when we manually added the vhdl files, and Keenan and I tried a bunch of stuff, so when that wasn't working we went to look for the build script the distributor includes with their design and this is what we found. The link if above if you would like to check it out.

reillymck, 2023-06-06:
I will be in tomorrow if you would like to get together and check it out also.

jgoeders, 2023-06-07:
Now that I think more about it, I'm pretty sure we can get the existing files to work with Vivado.  The source files originally were from Vivado projects that our students created several years ago for this work (https://ieeexplore.ieee.org/abstract/document/8494874).  This is where the `base` and `ooc` designs in BFASST come from.  

I then took these source files and organized them into BFASST directories and got them to work with Lattice.  So these same source files in the past have been successfully used in Vivado and Lattice.

@KeenanRileyFaulkner Perhaps you can try making a Vivado project manually with the files and see exactly what's preventing it from synthesizing?

KeenanRileyFaulkner, 2023-06-07:
I worked with Dallin on this for a bit and it looks like the problem is simply that the script generates an absolute file path for the library name, which should just be the basename. I'll change the script and see if I can get the normal xilinx_ooc flow to work with things that way, then let you know

KeenanRileyFaulkner, 2023-06-07:
Closed by #201

closed 2023-06-07 by KeenanRileyFaulkner