byuccl/bfasst #222: Format CI job should actually format PR

KeenanRileyFaulkner, 2023-06-16:
The format CI job should actually commit the formatted result instead of failing and making the user manually commit it. Should run on PR and run before any other jobs. Maybe have a "format bot?" Then the other jobs run after the format bot's commit.

jgoeders, 2023-06-16:
The challenge with this is is messes up the git blame...  

You could set up a git pre-commit or pre-push hook that checked formatting to remind the user.  

However, I have these set up in my VS Code user settings:

```json
    "[python]": {
        "editor.formatOnSave": true,
        "editor.defaultFormatter": "ms-python.black-formatter",
        "editor.formatOnType": true,
    },
    "python.formatting.provider": "black",
    "python.formatting.blackArgs": [
        "-l",
        "100"
    ],
```

If you have this set up then things will just always be formatted and you won't have to remember to run anything manually...

reillymck, 2023-06-22:
Yes and the settings.json in the bfasst project has format with black on save. For some reason though, vscode doesn't automatically stick with that? I know Keenan was having to manually format, so vscode was overriding this somewhere. I guess we could just say that the formatting is a personal issue.

closed 2023-06-22 by reillymck