byuccl/bfasst #246: Unit tests fail to clone github repo

reillymck, 2023-07-15:
Looks like there is something wrong with the servers' attempt to fetch the github repo.
```
Run actions/checkout@v3
Syncing repository: byuccl/bfasst
Getting Git version info
Copying '/home/jgoeders/.gitconfig' to '/home/jgoeders/actions-runner/_work/_temp/867c0907-9340-4817-9864-dbf05862306d/.gitconfig'
Temporarily overriding HOME='/home/jgoeders/actions-runner/_work/_temp/867c0907-9340-4817-9864-dbf05862306d' before making global git config changes
Adding repository directory to the temporary git global config as a safe directory
/usr/bin/git config --global --add safe.directory /home/jgoeders/actions-runner/_work/bfasst/bfasst
Deleting the contents of '/home/jgoeders/actions-runner/_work/bfasst/bfasst'
Initializing the repository
Disabling automatic garbage collection
Setting up auth
Fetching the repository
  /usr/bin/git -c protocol.version=2 fetch --no-tags --prune --progress --no-recurse-submodules --depth=1 origin +dacbd25c9d3d6dfd49c7c54fe4dce27b0d269625:refs/remotes/pull/245/merge
  remote: Duplicate header: "Authorization"
  Error: fatal: unable to access 'https://github.com/byuccl/bfasst/': The requested URL returned error: 400
  The process '/usr/bin/git' failed with exit code 128
  Waiting 18 seconds before trying again
  /usr/bin/git -c protocol.version=2 fetch --no-tags --prune --progress --no-recurse-submodules --depth=1 origin +dacbd25c9d3d6dfd49c7c54fe4dce27b0d269625:refs/remotes/pull/245/merge
  remote: Duplicate header: "Authorization"
  Error: fatal: unable to access 'https://github.com/byuccl/bfasst/': The requested URL returned error: 400
  The process '/usr/bin/git' failed with exit code 128
  Waiting [13](https://github.com/byuccl/bfasst/actions/runs/5550758485/jobs/10156901822?pr=245#step:3:14) seconds before trying again
  /usr/bin/git -c protocol.version=2 fetch --no-tags --prune --progress --no-recurse-submodules --depth=1 origin +dacbd25c9d3d6dfd49c7c54fe4dce27b0d269625:refs/remotes/pull/2[45](https://github.com/byuccl/bfasst/actions/runs/5550758485/jobs/10156901822?pr=245#step:3:50)/merge
  remote: Duplicate header: "Authorization"
  Error: fatal: unable to access 'https://github.com/byuccl/bfasst/': The requested URL returned error: 400
  Error: The process '/usr/bin/git' failed with exit code 128
```
This issue looks similar: https://github.com/actions/checkout/issues/838

jgoeders, 2023-07-18:
The .gitconfig on goeders0 is a symlink which is causing issues with the local runner (https://github.com/actions/checkout/issues/650).  I'll have to remove the symlink.

jgoeders, 2023-07-18:
.gitconfig on goeders0 is no longer a symlink.

closed 2023-07-18 by jgoeders