byuccl/bfasst #253: run_design.py doesn't print to stdout

jgoeders, 2023-07-21:
There is a `--quiet` option, but even when this isn't set, the command produces no output while it is running.  Is there a reason for this?  The old behavior printed the stdout of each tool.

jgoeders, 2023-07-22:
@KeenanRileyFaulkner

KeenanRileyFaulkner, 2023-07-22:
We got rid of the tools outputting to stdout because in the context of run_experiment it was ignored and with jobs in parallel it was a nightmare trying to debug pickling and output errors while passing a print lock all over the place. I have a few potential ideas for a fix:
* Make run design output the same thing as run experiment (aka a timer update)
* Pass the print lock around everywhere

I've started working on getting Dallin's bfasster ideas going and one of the things he's proposed is separating the creating of the ninja file from the running of it entirely. This would mean that you'd run bfasster to create the ninja file and then run ninja to execute it, both manually, from the command line. We could of course tie those together in a top level class/application runner, but using ninja in general will make this issue with stdout irrelevant. There will no longer be a run design and run experiment.

KeenanRileyFaulkner, 2023-09-21:
Now that we've gotten ninja merged, should this issue be closed @jgoeders ?

closed 2023-09-21 by jgoeders