byuccl/bfasst #295: Tcl_RegisterChannel: duplicate channel names

jgoeders, 2023-09-26:
The action runners occasionally encounter this issue:

https://github.com/byuccl/bfasst/actions/runs/6300725750/job/17104193811?pr=294#step:5:83

jgoeders, 2023-11-20:
Maybe try sourcing the vivado setup script (/tools/Xilinx/Vivado/2022.2/settings64.sh) before running Vivado?

jgoeders, 2023-11-22:
@dallinjdahl @KeenanRileyFaulkner @yonnorc42 

I did actually reproduce this locally today
```bash
(.venv) jgoeders@goeders-laptop:~/bfasst$ python scripts/bfasster.py --yaml experiments/tests/vivado_phys_netlist_cmp.yaml 
[65/176] vivado /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/impl.tcl
FAILED: /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/viv_impl.v /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/viv_impl.edf /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/impl.dcp /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/riscvSimpleDatapath.bit /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/utilization.txt 
export tempjou=$(mktemp); export templog=$(mktemp); /tools/Xilinx/Vivado/2022.2/bin/vivado -mode batch -journal $tempjou -log $templog -source /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/impl.tcl >&- && mv $tempjou /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/vivado.jou && mv $templog /home/jgoeders/bfasst/build/byu/riscvSimpleDatapath/in_context/impl/vivado.log
Tcl_RegisterChannel: duplicate channel names
[67/176] vivado /home/jgoeders/bfasst/build/byu/regfile/in_context/synth/synth.tcl
```

jgoeders, 2023-11-22:
Interestingly after killing and restarting it happened twice more right away:

```bash
KeyboardInterrupt
^C
(.venv) jgoeders@goeders-laptop:~/bfasst$ 
(.venv) jgoeders@goeders-laptop:~/bfasst$ python scripts/bfasster.py --yaml experiments/tests/vivado_phys_netlist_cmp.yaml 
[4/109] vivado /home/jgoeders/bfasst/build/byu/rx/in_context/impl/impl.tcl
FAILED: /home/jgoeders/bfasst/build/byu/rx/in_context/impl/viv_impl.v /home/jgoeders/bfasst/build/byu/rx/in_context/impl/viv_impl.edf /home/jgoeders/bfasst/build/byu/rx/in_context/impl/impl.dcp /home/jgoeders/bfasst/build/byu/rx/in_context/impl/rx.bit /home/jgoeders/bfasst/build/byu/rx/in_context/impl/utilization.txt 
export tempjou=$(mktemp); export templog=$(mktemp); /tools/Xilinx/Vivado/2022.2/bin/vivado -mode batch -journal $tempjou -log $templog -source /home/jgoeders/bfasst/build/byu/rx/in_context/impl/impl.tcl >&- && mv $tempjou /home/jgoeders/bfasst/build/byu/rx/in_context/impl/vivado.jou && mv $templog /home/jgoeders/bfasst/build/byu/rx/in_context/impl/vivado.log
Tcl_RegisterChannel: duplicate channel names
[5/109] vivado /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/impl.tcl
FAILED: /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/viv_impl.v /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/viv_impl.edf /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/impl.dcp /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/lightWave.bit /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/utilization.txt 
export tempjou=$(mktemp); export templog=$(mktemp); /tools/Xilinx/Vivado/2022.2/bin/vivado -mode batch -journal $tempjou -log $templog -source /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/impl.tcl >&- && mv $tempjou /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/vivado.jou && mv $templog /home/jgoeders/bfasst/build/byu/lightWave/in_context/impl/vivado.log
Tcl_RegisterChannel: duplicate channel names
```

KeenanRileyFaulkner, 2023-11-22:
Good to know. @yonnorc42 has to fix the path to source the settings script and then his pr will be ready for merge. If his fix doesn't work, we can investigate the relationship between killing the script and getting the error

jgoeders, 2023-11-27:
In ninja file (vivado rules), do something like:

`retry vivado ...`

PR #379 (2023-11-29): added retry to vivado rule to retry when tcl register fails
fixes #295﻿
closed 2023-11-29 by jgoeders