byuccl/bfasst #311: ninja_flow_manager.py doesn't find correct design path

jgoeders, 2023-10-09:
I try and run the ninja flow on the byu/tx design:

```python
(.venv) jgoeders@goeders-office:~/bfasst$ python bfasst/ninja_flows/ninja_flow_manager.py --flow vivado --designs designs/byu/tx/
```

but it errors looking for the design in the wrong place:

```python
Traceback (most recent call last):
  File "/home/jgoeders/bfasst/bfasst/ninja_flows/ninja_flow_manager.py", line 94, in <module>
    flow_manager.create_flows(
  File "/home/jgoeders/bfasst/bfasst/ninja_flows/ninja_flow_manager.py", line 35, in create_flows
    flow = get_flow(flow_name)(design, flow_args)
  File "/home/jgoeders/bfasst/bfasst/ninja_flows/vivado.py", line 12, in __init__
    self.vivado_tool = self.configure_vivado_tool(design, flow_args, ooc)
  File "/home/jgoeders/bfasst/bfasst/ninja_flows/flow.py", line 30, in configure_vivado_tool
    return Vivado(design, ooc=ooc)
  File "/home/jgoeders/bfasst/bfasst/ninja_tools/vivado/vivado.py", line 36, in __init__
    self.top = YamlParser(self.design / "design.yaml").parse_top_module()
  File "/home/jgoeders/bfasst/bfasst/yaml_parser.py", line 103, in parse_top_module
    self.__read_experiment_yaml()
  File "/home/jgoeders/bfasst/bfasst/yaml_parser.py", line 35, in __read_experiment_yaml
    with open(self.yaml_path) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jgoeders/bfasst/designs/tx/design.yaml'
```

KeenanRileyFaulkner, 2023-10-09:
We are a little constrained on the design paths at the moment and the yaml parser probably needs a new approach to fix it. Currently, specifiying the sub-path from bfasst/designs will work. So if you ran the command as 
`python bfasst/ninja_flows/ninja_flow_manager.py --flow vivado --designs byu/tx/`
it would work.

KeenanRileyFaulkner, 2023-10-09:
Please note that the ninja_flow_manager and flows/tools in general are intended to create ninja snippets. If you want the snippets automatically run as part of the program, you would need to call the script `python scripts/bfasster.py` rather than directly invoking the flow manager. If you do not want them run automatically, you can manually run them after invoking the flow manager by running `ninja` in the same directory in which `build.ninja` is created

jgoeders, 2023-10-09:
@KeenanRileyFaulkner OK, I wasn't sure what file to invoke first.  I'll check if this problem also exists when invoking bfasster.py

KeenanRileyFaulkner, 2023-10-09:
@jgoeders this issue primarily exists because we focused on supporting the syntax of our current CI yaml files and how they specify design paths over supporting specifiying a full relative path name for each design. More elegant solutions are almost certainly out there. yaml_parser.py needs to be fixed. So this issue can stay open, but in the meantime, the project is still working. You just have to start design paths with the subdir from `designs/`, rather than starting them with `designs/` itself.

PR #312 (2023-10-10): Update ninja tools to handle relative paths
This updates the ninja system to do a few things:
* Expand all paths to absolute paths before passing them to the tools 
* Update the tools to expect this absolute path and create the build directories using relative_to (Fixed #311)
* Update the `Flow` class to track the design.  We are only using this in one place, but it might be useful in the future
* Update bfasster.py to not pass argument dictionaries between the functions and to better provide error messages on incorrect usages.
* Add pylint ignore to fix #306
closed 2023-10-10 by jgoeders