byuccl/bfasst #336: Add implicit dependencies for python snippets used by rules in ninja build snippet templates

KeenanRileyFaulkner, 2023-11-07:
If a ninja build snippet will cause a python script to be invoked (for example, one [in ninja_utils](https://github.com/byuccl/bfasst/tree/main/bfasst/ninja_utils) or anything in third_party), the script file path should be an implicit dependency of the rule as [seen here](https://github.com/byuccl/bfasst/blob/169c8f71b405cd33d697a1dfcffde3259e85e1c3/bfasst/ninja_tools/transform/netlist_cleanup_build.ninja.mustache#L1C96-L1C138).

jgoeders, 2023-11-26:
@KeenanRileyFaulkner When should a python be script be placed as a dependency in the ninja rule versus being added by `add_ninja_deps`?  Should it be in both places?  Or just one of them?

KeenanRileyFaulkner, 2023-11-26:
The ninja deps have to do with rebuilding the build.ninja file when they change. The implicit dependencies will cause other snippets to be rerun. So if a Python script should cause the build file to be recreated (which I think is almost always true), it should be included in the add ninja deps method. But I also think that a Python script should almost always be an implicit dependency of the build snippet that calls it so that the snippet reruns if it changes.

jgoeders, 2023-11-26:
> ... cause the build file to be recreated (which I think is almost always true)...but I also think that a Python script should almost always be an implicit dependency of the build snippet that calls it so that the snippet reruns if it changes.

Does recreating the build file invalidate all output products and cause everything to be re-run?  (in which case the latter wouldn't be required)

KeenanRileyFaulkner, 2023-11-26:
No, it doesn't invalidate outputs unless the build snippet that created the output has dependencies that change. This gets checked when the build file is redone and executed. If none of the dependencies for rules other than the configure rule change, there is still no work to do.