byuccl/bfasst #341: CI brings in all submodules, but often doesn't need to

jgoeders, 2023-11-13:
Can we bring in fewer submodules in the CI when they are not needed?

Perhaps using this approach:
```
git submodule init <submodule a> <submodule b>
```

yonnorc42, 2023-11-29:
@jgoeders Would this need to be approached differently because of some of the changes that you've made recently?

jgoeders, 2023-11-29:
@yonnorc42 

I didn't really have a plan for this.  Different tests require different tools.  For example, I think only one tool requires yosys, but we spend quite a bit of time setting it up for every test.  We don't have a mechanism for conveying what tools are needed from the test up to the environment setup.  I don't have any good ideas on how to do this.  Maybe others have thoughts? @dallinjdahl @KeenanRileyFaulkner @reillymck

reillymck, 2023-11-29:
I was thinking we could make recipes for different unit tests and instead of make install it makes the test-specific recipe.

jgoeders, 2023-12-01:
I was thinking more along the lines of having something like this in the yaml files:

```yaml
tools: vivado yosys rapidwright
```

And then the environment setup code could iterate through these and call `make install_yosys`, `make install_rapidwright`, etc.

jgoeders, 2023-12-02:
I've started working on this in the linked PR; however, I think the correct way to do this would be to have the tools listed in the yaml that details all flows (discussed in #362 and #382), rather than in the test yaml file.

KeenanRileyFaulkner, 2023-12-02:
I've been working on that yaml file, and will add this functionality.

closed 2023-12-17 by jgoeders