byuccl/bfasst #351: Weekly tests failing - need access to gmt_tools

jgoeders, 2023-11-20:


jgoeders, 2023-11-20:
https://github.com/byuccl/bfasst/blob/main/.github/workflows/unittests.yml#L49

KeenanRileyFaulkner, 2023-11-28:
This issue is not resolved. See [weekly tests run 25](https://github.com/byuccl/bfasst/actions/runs/6992178706)

yonnorc42, 2023-11-28:
That pull request should fix it. I had to add the private gmt_tools key in another location I missed the first time.

KeenanRileyFaulkner, 2023-12-04:
This is still an issue

yonnorc42, 2023-12-27:
Weekly tests are still failing because of failing to clone gmt_tools on the runners @jgoeders

jgoeders, 2024-01-10:
This needs to be removed: https://github.com/byuccl/bfasst/blob/main/.github/workflows/weeklytests.yml#L45

And the specific tools that are needed will need to be installed like [here](https://github.com/byuccl/bfasst/blob/main/.github/workflows/unittests.yml#L98) or [here](https://github.com/byuccl/bfasst/blob/main/.github/workflows/unittests.yml#L56)

PR #353 (2023-11-21): gmt tools
Fixes #351
- added gmt_tools key to weekly_tests
PR #377 (2023-11-28): Fix gmt_tools access
This should fix #351
PR #431 (2024-01-23): pass weeklytests
Removed designs from some of the weeklytests because they have issues. Reduced the number of runs on error_injection because it was timing out. 

The commented out designs in the weeklytests yaml's are what was causing the weeklytests to fail. 

The tests are also a bit inconsistent on whether they pass or not because of the 40 minute timeout limit. Sometimes a test will finish in 15 minutes and pass, and then sometimes in 40 minute it won't finish, and will timeout. I'm not sure what causes this. I could increase the timeout length but I feel like a test deserves to fail if i [... cut]
closed 2024-01-23 by jgoeders