byuccl/bfasst #362: Flow names on command-line

jgoeders, 2023-11-26:
I noticed that when running flows from the command-line the code converts the class name from PascalCase to snake_case.  Is this done to keep lower-case command-line convention?  In my opinion it might be more clear to just have the command-line name match the class name, even if it means we are case-insensitive.  Thoughts?

KeenanRileyFaulkner, 2023-11-30:
The one problem with this is how we obtain access to each flow class at the moment. We use the following utility method in the flow manager, which has to import the module before it gets the flow class.

```
def get_flow(flow_name):
    """Get a flow by name"""
    flow_module = import_module(f"bfasst.ninja_flows.{flow_name}")
    flow_class = "".join(word.capitalize() for word in flow_name.split("_"))
    return getattr(flow_module, flow_class)
```

If we were to switch to specifying the name of the class instead of the name of the module, we would have to import all the flow modules and search each one until we find the one that contains the class we specified. Is that what we want to do?

jgoeders, 2023-11-30:
Oh I didn't realize it was based on module name.

How are we loading the list for the argparse `choices`?

KeenanRileyFaulkner, 2023-12-01:
It appears that `choices` was removed when `bfasster.py` moved to `run.py`. However, it was loaded using [this other flow utility function](https://github.com/byuccl/bfasst/blob/1a9be2155d6ad6f56204fec5e114c45d88ce5f80/bfasst/ninja_flows/flow_utils.py#L8), which returns the names of all the flow modules.

jgoeders, 2023-12-01:
I think it would be ok to have a single file (perhaps yaml) where we keep a list of flows and their description, which would be used for #382.

KeenanRileyFaulkner, 2023-12-01:
Ok, but we'd keep the same approach for accessing them, right? Or we want to change to import all the modules and find the correct flow class?

closed 2023-12-04 by jgoeders