byuccl/bfasst #387: VivadoSynthFromTcl tool is redundant

KeenanRileyFaulkner, 2023-12-02:
The Vivado located in `bfasst/tools/synth/vivado_synth.py` is sufficient for doing what the VivadoSynthFromTcl tool is doing. If you compare the following two files, you will see they are essentially doing the same thing when they call the vivado rule.

[Generic vivado build snippet](https://github.com/byuccl/bfasst/blob/main/bfasst/tools/synth/vivado_synth_build.ninja.mustache)
[Redundant vivado build snippet](https://github.com/byuccl/bfasst/blob/main/bfasst/tools/synth/vivado_synth_tcl_build.ninja.mustache)

We could work to remove the VivadoSynthFromTcl class from the project (currently used only by the RandSoc flow) by simply placing a conditional mustach syntax around the build synth.tcl snippet in the file like this, and then updating the synth tool accordingly:

```
{{#create_tcl}}
build {{ synth_output }}/synth.tcl: template {{ synth_output }}/synth.json {{ synth_library }}/vivado_synth.tcl.mustache
{{/create_tcl}}

{{#in_context}}
build {{ synth_output }}/{{ top }}.xdc: vivado_ioparse {{ synth_output }}/report_io.txt
{{/in_context}}

build {{ synth_output }}/viv_synth.edf {{ synth_output }}/synth.dcp {{#in_context}}{{ synth_output }}/report_io.txt{{/in_context}}: vivado {{ synth_output }}/synth.tcl | {{#verilog}}{{ . }} {{/verilog}} {{#system_verilog}}{{ . }} {{/system_verilog}}
    journal = {{ synth_output }}/vivado.jou
    log = {{ synth_output }}/vivado.log
    cwd = {{ cwd }}
```

As it stands, having both tools is a bit confusing since Vivado Synth already uses a tcl file to operate.

yonnorc42, 2024-04-14:
@KeenanRileyFaulkner @jgoeders Do we want to implement this or should we close it?

reillymck, 2024-04-14:
I have made this obsolete in my branch. I will work on getting that into main. @yonnorc42

closed 2024-06-10 by yonnorc42