byuccl/bfasst #389: Update unit tests

KeenanRileyFaulkner, 2023-12-02:
* Files in `test/` dir need to be updated to ensure coverage of recently added flows
* Considerations may need to be made for runtime on some of them
* We could also combine the `test/` and `tests/` directories safely, and eliminate old yaml files that have become legacy/unrunable

jgoeders, 2023-12-02:
I'm not sure I want to combine test and tests, as they mean somewhat different things.  `test` is specifically for unit tests, while `tests` because a shortened form of experiment configurations.  When I suggested the `tests` name to @yonnorc42, the closeness with `test` didn't occur to me.  We probably need a better name for these files.  `configs`?

KeenanRileyFaulkner, 2023-12-02:
That's fine. I made the suggestion because I think the unittest module will support either test or tests as the name of the folder and those tests have also integrated into CI. But we can keep them separate and rename the one with the yaml files.

jgoeders, 2023-12-04:
Lets move all yaml files to test/yml/* and remove the non-ci related configurations.

yonnorc42, 2023-12-14:
@jgoeders So do you want to move all the `weeklytests` and `ci` yaml files to the `test/` directory? And then get rid of all the yaml files in `tests/` that aren't `weeklytests` or `ci`?

PR #430 (2024-01-16): add rand_soc unittest
- desin -> design typo fixed in one of the test files
- Added testing for rand_soc flow
- Other small test adjustments
Should we delete all the extra .yaml configurations in the tests/ directory?
Fixes #389
closed 2024-01-16 by jgoeders