byuccl/bfasst #399: Towards removal of scripts/run_design.py and scripts/run_experiment.py

KeenanRileyFaulkner, 2023-12-06:
It would be nice to complete the transition of all legacy_flows to the new architecture using ninja, and then to remove the old run scripts, flows, and tools from the project. Priority was given to Reilly's tools and flows, but now that all of them are moved over, this could be more doable in the coming semester.

I would propose this transition is done in a set of smaller PRs, where each one is designed to transition over a single flow, and any necessary tools related to it.

KeenanRileyFaulkner, 2023-12-19:
@jgoeders it looks like we recently removed run_design.py and run_experiment.py. Does this mean we are not going to support the above legacy flows any more and should just delete them too? It's difficult for me to transition these flows over without being able to run them on the old architecture to see how they work.

jgoeders, 2023-12-19:
Let's leave the files for now.  I would like to try and migrate those that we can.  We can always check out an old commit of bfasst to compare how they run.

I assume we would need to get the icecube tools installed to run most of these flows and try them.

yonnorc42, 2023-12-21:
@KeenanRileyFaulkner run_design.py and run_experiment.py are still there. scripts/scripts_legacy/

yonnorc42, 2024-05-21:
@jgoeders I was looking at moving the gather_impl_data flow over to the new architecture but I'm not sure how it would work. For example, during the flow, it runs the icestorm reverse bitstream tool on netlists created from different synthesis tools, although they all come from the same design. Because of how ninja works, doesn't this mean that all the reversed netlists, except for the last one created, would be trampled?

jgoeders, 2024-05-22:
By the name, it sounds like gather_impl_data is just gathering stats from the implementation of runs?  If so, we can probably just skip this as we have newer ways of doing this.  Since we aren't collecting data using the old legacy tools, this should be fine.