byuccl/bfasst #402: Update make pylint target

KeenanRileyFaulkner, 2023-12-07:
Currently, the `make pylint` target checks all `.py` files in HEAD that differ from the common ancestor of HEAD and origin/main. This kind of approach is a relic of ensuring that the codebase need not be refactored all at once when pylint was first added to it. 

The CI checks handle pylint differently; they simply check all `.py` files at HEAD in the `scripts/` and `bfasst/` directories using the following command:
```
pylint --errors-only $(git ls-files --directory scripts --directory bfasst | grep -e ".py$")
```

The downside of the current approach for the `make pylint` target is that it will error out if files are deleted from the working branch that still exist in main. Because the project has largely been made to follow pylint requirements at this point, it could be helpful to switch the `make pylint` target to use the same approach as the CI checks.

yonnorc42, 2023-12-08:
@KeenanRileyFaulkner How do you feel about this?

```
pylint: format
	git fetch
	pylint --errors-only $$(git ls-files --directory scripts --directory bfasst | grep -e ".py$$")
	pylint $$(git diff --name-only | grep -e ".py$$")
```

KeenanRileyFaulkner, 2023-12-08:
Looks fine to me, assuming it doesn't miss anything locally (since it won't impact CI at all).

PR #401 (2023-12-13): create legacy tools
- Moved legacy tools to new directory tools_legacy
- Added new subdirectory of scripts called scripts_legacy 
- Changed pylint to no longer compare to origin/main
- Added new flow VivadoBitToNetlist
- Fixes #315 
- Fixes #402
closed 2023-12-13 by jgoeders