byuccl/bfasst #408: vivado invocation logs

jgoeders, 2023-12-12:
In our vivado [rules](https://github.com/byuccl/bfasst/blob/main/bfasst/tools/common/vivado_rules.ninja.mustache), the command is:

```
command = export tempjou=$$(mktemp); export templog=$$(mktemp); cd $cwd && {{ utils_path }}/retry.sh {{ vivado_path }} -mode batch -journal $$tempjou -log $$templog -source $in >&- && mv $$tempjou $journal && mv $$templog $log
```

I'm wondering why we are putting the logs in a temp directory. It's a bit annoying because you can't look at them while they are running.

Also the logs are only copied over if the tool runs correctly...

jgoeders, 2023-12-16:
@KeenanRileyFaulkner

KeenanRileyFaulkner, 2023-12-18:
We are doing this because we had issues at one point with multiple instances of vivado trampling each other's logs since they were all named the same thing. I don't know if that's still a problem, given the current setup of our build directory. As far as making sure they move over even if the command fails, can't we just change the last two `&&` symbols into `;`?

jgoeders, 2023-12-19:
OK, I did some testing without making a temporary file and I didn't get any trampling, but it did create a bunch of temporary "backup" files that were annoying.  However, I found a way to disable them.  https://support.xilinx.com/s/question/0D52E00006iI3HUSA0/how-to-disable-backup-log-and-jou-files?language=en_US

KeenanRileyFaulkner, 2023-12-19:
The link above seems to suggest that you just disable the logs from being created at all? Or are you referring to the comment about setting the parameter to 0 in the init tcl file?

KeenanRileyFaulkner, 2023-12-19:
Where is the init tcl file created?

jgoeders, 2023-12-19:
You have to create it yourself in `$HOME/.Xilinx/Vivado/init.tcl`

KeenanRileyFaulkner, 2023-12-19:
Oh, ok. Now that I think about it, this is probably the main reason we were using temp files for the logs and moving them. I can just revert that, though, and the user can turn the backup files off on their own.

KeenanRileyFaulkner, 2023-12-19:
An alternative to this is to change the `&&` chains on the vivado rule to `;`. This would make it so that the log is still created as a temp file, but is moved over to the correct location even if the tool fails. You still wouldn't be able to watch the log as it is written, but you also would not have to deal with backup logs.

PR #411 (2024-01-12): update unittests
- Added test for VivadoBitToNetlist flow
- unittest no longer runs any of the flows end-to-end
- run-test now checks if ninja has more work to do after the flow runs (except on vivado_phys_netlist_error_inject.yaml)
- vivado_ooc weeklytest now excludes 2 designs in ooc/ because "Some designs in the OOC directory have too many I/Os to be placed"
- vivado logs aren't put into temp directory and then moved over if tool succeeds anymore
- Adjusted quick_test_flows.sh (does anyone use this?)
Fixes #421 
Fixes #408
closed 2024-01-12 by jgoeders