byuccl/bfasst #416: Can't reproduce unittest error.

jgoeders, 2023-12-17:
I can't reproduce this error.  I've tried on a fresh install

https://github.com/byuccl/bfasst/actions/runs/7235847371/job/19713756024?pr=385

I suspect this is another case of interaction between the tools in the build directory since the unittest runs them all back-to-back.

jgoeders, 2023-12-17:
@KeenanRileyFaulkner Unittest is failing in the CI, but I can't reproduce.  Can you reproduce the issue?

KeenanRileyFaulkner, 2023-12-18:
Given that the thing that failed is actually running VivadoBitAnalysis, I'm guessing that it failed due to private key/gh secret stuff, which would be difficult to reproduce locally. I don't know if it's an issue with not even trying to install the third party libraries, or if it is an issue with failure to authenticate

jgoeders, 2023-12-18:
Good catch.  It did fail to clone the repo.  https://github.com/byuccl/bfasst/actions/runs/7235847371/job/19713756024?pr=385#step:5:5292

I'm not sure why it didn't error and stop there.

closed 2023-12-18 by jgoeders