byuccl/bfasst #421: Unittest has long run time.  Move long running tests to existing tests.

jgoeders, 2023-12-18:
Unittest takes a long time to run.  This is mainly due to the tests that run the flows, then invoke ninja again to make sure there's no work to do.  This is a good test, but we could move this extra check to our other tests that run the flows.  At the end of running the CI yaml file, it could have another step that runs ninja to make sure there's no work to do.

yonnorc42, 2023-12-21:
@jgoeders I'm a little bit confused by this. Isn't the reason that unittest takes so long is because some of the flows like VivadoBitAnalysis just take a long time to run? Does invoking ninja again really affect the runtime that much?

jgoeders, 2023-12-21:
[image]

The idea is that we don't need unittests (the first one in that pic) to run the full end-to-end flows, because they are already being run by the "run-test" tests.  So we can remove those end-to-end tests from unittests.  However, the functionality that checks that ninja doesn't have any more work to do is a nice check, but we could move it to the end of the "run-test" CI.

Does that make sense?

yonnorc42, 2023-12-21:
Yes that does make sense. So would you add this functionality in the unittests.yml file under the run-test job?

jgoeders, 2023-12-22:
Yes, exactly.  I'm not sure what that would look like.  If you can't do it from a set of bash commands, you may need a separate python script.

yonnorc42, 2023-12-23:
@jgoeders I added this into the update unittests pull request. Let me know if you want me to change how it's implemented.

PR #411 (2024-01-12): update unittests
- Added test for VivadoBitToNetlist flow
- unittest no longer runs any of the flows end-to-end
- run-test now checks if ninja has more work to do after the flow runs (except on vivado_phys_netlist_error_inject.yaml)
- vivado_ooc weeklytest now excludes 2 designs in ooc/ because "Some designs in the OOC directory have too many I/Os to be placed"
- vivado logs aren't put into temp directory and then moved over if tool succeeds anymore
- Adjusted quick_test_flows.sh (does anyone use this?)
Fixes #421 
Fixes #408
closed 2024-01-12 by jgoeders