byuccl/bfasst #444: vivado_phys_netlist_error_inject not deterministic

jgoeders, 2024-02-07:
I'm guessing it's not being seeded....

yonnorc42, 2024-02-29:
@jgoeders I believe it is being seeded. I checked the log files for the bit/wire swaps and the same bit/wire is being flipped each time the flow runs.

jgoeders, 2024-03-04:
@yonnorc42 I thought I had linked a case where it was causing the CI to fail.  Do you remember where that was? Can you add the link here?  

I restarted the CI and it passed, which is why I was convinced something was non-deterministic.

yonnorc42, 2024-03-05:
@jgoeders https://github.com/byuccl/bfasst/actions/runs/7790517471/job/21244555884?pr=443

The weird part is that I checked and it is being seeded. But on occasion a different wire is flipped anyway. I haven't looked very deep into it yet so I'm not sure why.