byuccl/bfasst #451: Pass -j to ninja

jgoeders, 2024-03-01:
The run.py should have a way to pass the '-j' argument to ninja.  Sometimes you need to do a serial run...

KeenanRileyFaulkner, 2024-04-04:
There should never need to be a reason to pass `-j 1` to ninja because any jobs in a serial run should have a dependency (perhaps an implicit dependency using the pipe character) on the job that comes directly before it. In general, I think it's best to parallelize as much as can be by allowing ninja to default and match the number of threads it uses to the number of cores, and then use implicit dependencies in the ninja build snippets to enforce serial order of things as necessary. @jgoeders

jgoeders, 2024-04-10:
Sometimes you need to serialize when collecting results.  For example, when writing a paper and reporting the speed to run different passes, it's a bit nicer to collect results on a single core without the machine loaded down.

PR #467 (2024-04-10): ninja_mode
- Added the option to run ninja in serial mode
Fixes #451
closed 2024-04-10 by jgoeders