byuccl/bfasst #476: Redundant unittests?

reillymck, 2024-04-30:
https://github.com/byuccl/bfasst/blob/main/test/flows/test_vivado_flow.py#L46
https://github.com/byuccl/bfasst/blob/main/test/scripts/test_ninja_flow_manager.py#L95

These two seem to be doing the same thing (and the same for every file in test/flows). Maybe I'm missing something?

reillymck, 2024-04-30:
Both files are being ran though. Total overlap between them should be investigated.

yonnorc42, 2024-05-09:
Yeah I think you're right, they seem to do the same thing. But for the same flow, Vivado, why does one of them have the expected number of build statements at 5, and for the other, 6?

reillymck, 2024-05-09:
I'm not sure, but as I've been running the unit tests for my pr, it seems that not all of the redundant ones run. It looked like possible hanging code.

PR #479 (2024-05-09): remove redundant unittests
Fixes #476
closed 2024-05-09 by reillymck