byuccl/bfasst #488: randsoc_dump dependency

jgoeders, 2024-06-11:
There seems to be a missing dependency causing randsoc_dump to fail sometimes.

This will create `design_0/design.tcl`:
```
build /home/jgoeders/bfasst/build/rand_soc/design_0/design.tcl: rand_soc | /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/paths.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/utils.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ports.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/creator.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/__init__.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/emc.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/slice_and_concat.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/accumulator.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/ip_base.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/gpio.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/axi.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/uartlite.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/microblaze.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/intc.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/rand_soc/ip/clk_gen.py /home/jgoeders/bfasst/third_party/gmt_tools/rand_soc/main.py
    design_dir_path = /home/jgoeders/bfasst/build/rand_soc/design_0
    seed = 0
    part = xc7a200tlffv1156-2L
```

However, the output file is not a dependency for the subsequent synthesis step:
```
build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/setup.tcl: template /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.json /home/jgoeders/bfasst/bfasst/tools/synth/vivado_synth_setup.tcl.mustache
build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.tcl: template /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.json /home/jgoeders/bfasst/bfasst/tools/synth/vivado_synth_run.tcl.mustache
build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/reports.tcl: template /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.json /home/jgoeders/bfasst/bfasst/tools/synth/vivado_synth_reports.tcl.mustache
build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/run.tcl: template /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.json /home/jgoeders/bfasst/bfasst/tools/common/vivado_top_tcl.mustache

build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/design.xdc: vivado_ioparse /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/report_io.txt

build /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/viv_synth.edf /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.dcp /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/report_io.txt: vivado /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/run.tcl |   /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/setup.tcl /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/synth.tcl /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/reports.tcl
    journal = /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/vivado.jou
    log = /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth/vivado.log
    cwd = /home/jgoeders/bfasst/build/rand_soc/design_0/vivado_synth
```

reillymck, 2024-06-11:
I think I have a good fix for this, let me push it up.

jgoeders, 2024-06-11:
Fixed by #489

closed 2024-06-11 by jgoeders